adafruit/Adafruit_CircuitPython_TCA9548A #8: Deadlock when connected i2c device fails.

StevenBruinen, 2019-04-03:
I've connected some Adafruit SHT31-D Temperature & Humidity Sensors to the TCA9548A but once in a while one of the sensors fails to read properly with a remote I/O error. This is probably caused by the long wiring but in my application it's not a big deal as the program just tries again.   

At least that was the case before I used the TCA9548A with just 2 sensors on the i2c of my Raspberry Pi. As the reading was done in a try: / except: the program continued to run. The remote I/O error didn't cause the i2c bus to hang so most of the time the next reading was succesful.

But when using the TCA9548A it's causing a deadlock on this error:

```
Traceback (most recent call last):
  File "/home/pi/MainProgram.py", line 332, in ReadTempValue
    TemperatureValue = '{0:4.1f}'.format(THsensor.temperature)
  File "/usr/local/lib/python3.5/dist-packages/adafruit_sht31d.py", line 120, in temperature
    raw_temperature, _ = self._data()
  File "/usr/local/lib/python3.5/dist-packages/adafruit_sht31d.py", line 109, in _data
    i2c.readinto(data)
  File "/usr/local/lib/python3.5/dist-packages/adafruit_bus_device/i2c_device.py", line 97, in readinto
    self.i2c.readfrom_into(self.device_address, buf, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/adafruit_tca9548a.py", line 76, in readfrom_into
    return self.tca.i2c.readfrom_into(address, buffer, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/busio.py", line 55, in readfrom_into
    return self._i2c.readfrom_into(address, buffer, stop=stop)
  File "/usr/local/lib/python3.5/dist-packages/adafruit_blinka/microcontroller/generic_linux/i2c.py", line 44, in readfrom_into
    readin = self._i2c_bus.read_bytes(address, end-start)
  File "/usr/local/lib/python3.5/dist-packages/Adafruit_PureIO/smbus.py", line 155, in read_bytes
    return self._device.read(number)
OSError: [Errno 121] Remote I/O error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/MainProgram.py", line 336, in ReadTempValue
    TemperatureValue = '{0:4.1f}'.format(THsensor.temperature)
  File "/usr/local/lib/python3.5/dist-packages/adafruit_sht31d.py", line 120, in temperature
    raw_temperature, _ = self._data()
  File "/usr/local/lib/python3.5/dist-packages/adafruit_sht31d.py", line 106, in _data
    self._command(SHT31_MEAS_HIGHREP)
  File "/usr/local/lib/python3.5/dist-packages/adafruit_sht31d.py", line 100, in _command
    with self.i2c_device as i2c:
  File "/usr/local/lib/python3.5/dist-packages/adafruit_bus_device/i2c_device.py", line 175, in __enter__
    while not self.i2c.try_lock():
  File "/usr/local/lib/python3.5/dist-packages/adafruit_tca9548a.py", line 65, in try_lock
    self.tca.i2c.writeto(self.tca.address, self.channel_switch)
  File "/usr/local/lib/python3.5/dist-packages/busio.py", line 65, in writeto
    return self._i2c.writeto(address, buffer, stop=stop)
  File "/usr/local/lib/python3.5/dist-packages/adafruit_blinka/microcontroller/generic_linux/i2c.py", line 38, in writeto
    self._i2c_bus.write_bytes(address, buffer[start:end])
  File "/usr/local/lib/python3.5/dist-packages/Adafruit_PureIO/smbus.py", line 244, in write_bytes
    self._device.write(buf)
OSError: [Errno 121] Remote I/O error
```

When I run a debug I see it's waiting for a lock on the i2c bus:

```
(gdb) py-list
  54        """An object that must be locked to prevent collisions on a microcontroller resource."""
  55        _locked = False
  56
  57        def try_lock(self):
  58            """Attempt to grab the lock. Return True on success, False if the lock is already taken."""
 >59            if self._locked:
  60                return False
  61            self._locked = True
  62            return True
  63
  64        def unlock(self):
 
```
 
It seems the busio never releases the lock on the bus when running this setup and an error occures. But as it does without the TCA I assume there is some problem with the error handling in the TCA library. 

Any ideas?

kattni, 2020-05-04:
@StevenBruinen Are you still experiencing this issue?

StevenBruinen, 2020-05-05:
Because of this issue I don't use the TCA anymore in my project anymore only the sensor which works fine. So for me its not an issue at the moment.

closed 2020-05-05 by StevenBruinen