adafruit/Adafruit_CircuitPython_TCA9548A #9: Support multiple TCAs

caternuson, 2019-06-26:
There's a simple way to disable the channel output for TCAs not currently in use:
https://learn.adafruit.com/adafruit-tca9548a-1-to-8-i2c-multiplexer-breakout/wiring-and-test#multiple-multplexers-4-13
but not sure how that could get implemented here and still maintain the current ease of use.
```python
tca1 = adafruit_tca9548a.TCA9548A(i2c, address=TCA1_ADDRESS)
tca2 = adafruit_tca9548a.TCA9548A(i2c, address=TCA2_ADDRESS)

sensor1 = SomeSensor(tca1[0])
#
# etc for sensors 2-8
#
sensor9 = SomeSensor(tca2[0])
#
# etc for sensors 10-16
#
```
Maybe end all transactions with a disable-all-channels?

tannewt, 2019-07-01:
Why is it needed when each mux has it's own address?

caternuson, 2019-07-02:
You only use the mux address to change the output channel state. After that, it's I2C as usual. So there's a potential to have a conflict with multiple devices of same address on different muxes. The fix is to turn off all channels on muxes not being used:
https://learn.adafruit.com/adafruit-tca9548a-1-to-8-i2c-multiplexer-breakout/wiring-and-test#multiple-multplexers-4-13
to "hide" the conflicting sensors.

I'm thinking this might just be a one liner in `unlock`:
https://github.com/adafruit/Adafruit_CircuitPython_TCA9548A/blob/master/adafruit_tca9548a.py#L68
that sends out the "disable all" command. Then the context manager will call that on exit:
https://github.com/adafruit/Adafruit_CircuitPython_BusDevice/blob/master/adafruit_bus_device/i2c_device.py#L164
Once you're done talking to a sensor, there's no reason to leave the output channel open.

tannewt, 2019-07-02:
Ah interesting. Ya, unlock sounds like the correct place.

caternuson, 2019-07-25:
Closing. Added with #10

closed 2019-07-25 by caternuson