adafruit/Adafruit_CircuitPython_TCA9548A #21: Add Mock Library to support development outside RPi

codenio, 2021-01-03:
Hi, adding a mock package in this same library, would allow people to develop their code outside RPi

I am currently using the following way to mock adafruit_tca9548a outside RPi, and it works.

```python
try:
    import RPi.GPIO as GPIO
    import adafruit_tca9548a
except:
    import Mock.GPIO as GPIO
    import adafruit_tca9548a.mock as adafruit_tca9548a
```

Thoughts..?

ladyada, 2021-01-03:
you should be able to use this package outside of raspberry pi, no need to use RPi.GPIO library at all, `blinka` handles just about every single board computer

codenio, 2021-01-03:
how about ubuntu ?
for development in normal computers where you dont have access to single board computers..

ladyada, 2021-01-03:
yes blinka works with desktop linux just fine if you have an MCP2221 or FT232H GPIO adapter - you _ could_ make a mock device for when there's no hardware at all but that would go into `blinka`
https://learn.adafruit.com/search?q=blinka

closed 2021-01-03 by ladyada

codenio, 2021-01-03:
in case if is to be added in blinka, then entire RPi needs to be mocked.

FYI, had done this trough [VPi](https://github.com/codenio/virtual-pi),
to support developement when there is no hardware available

ladyada, 2021-01-03:
that would be a path yes!