adafruit/Adafruit_CircuitPython_TCA9548A #24: read addresses of hardware on several multiplexers

dretful, 2021-02-22:
Hi.  I'm new to this so sorry if I have done it all wrong, and my general ignorance.

I have a pi pico and two tca9548a devices.  One on 0x70 and another on 0x72 

As I see it this means I have an extra 16 'slots' to attach address-contended (or indeed any) devices.

I have used `i2c.scan()` and can find the multiplexers, and any other hardware on the main bus.
I have then instantiated the 'found' multiplexers on the i2c bus ( I've assumed they are multiplexers die to their address, but can the library prove that they are?), using this:
`adafruit_tca9548a.TCA9548A(bus, address=hex(l))`
I have connected random hardware to the multiplexers in random places, and I want to 'find' each connected device on its channel by scanning the 8 possible channels for each of the two and see what's there (by getting an address if there is one, like with i2c.scan(), somehow). 
I think I can change the channel on a given mux using (tca[mux] )is the hex address)
`TCA9548A_Channel(tca[mux],channel)`

I cant see how to do this with the library? is it possible?

caternuson, 2021-02-22:
Hmmmm. Maybe we should add a pass thru for `scan()`? You would then need to use `try_lock()` as normal. It would look something like this:
```python
tca[3].try_lock()
tca[3].scan()
```

The TCA channel switch is buried in `try_lock()`:
https://github.com/adafruit/Adafruit_CircuitPython_TCA9548A/blob/9acbf9cc4722e6b150057c5eede188d65093f6d9/adafruit_tca9548a.py#L49
which is called automatically when using a context manager, which is the more regular use case. But you can also just call it directly.

dretful, 2021-02-23:
Thanks for the reply, I really appreciate it 

Before getting back into the scanning, I wanted to be sure the multiplexers were addressing devices correctly. So I plugged a bunch in and wrote the code.  I've attached it (tca example.py), and it works (apart from a problem reading from the second of two ads1115's that are directly on the I2C bus - but that's not important now, the multiplexer part works fine.

In the posted code I've created a list of multiplexers, and I used that to attach the devices to their ports.

So on to the scan problem and your example,  Assuming that means lock channel 3?

to test in its most basic form:
```
import board
import busio
import adafruit_tca9548a
i2c = busio.I2C(board.GP1, board.GP0)
tca=adafruit_tca9548a.TCA9548A(i2c, hex(0x70)) 
tca[3].try_lock()
```

  File "<stdin>", line 6, in <module>
  File "adafruit_tca9548a.py", line 49, in try_lock
TypeError: can't convert str to int

if I try
'tca.try_lock()'

then
  File "<stdin>", line 6, in <module>
AttributeError: 'TCA9548A' object has no attribute 'try_lock'

Really don't get it.  I'm trying, really I am!

FWIW I tried this in my scanner too:
I have two tca's 0x70 and 0x72.  They are in the list in positions tmux[0] and tmux[2]. that way I can refer to the particular device.
You said use 'tca[3].try_lock()'.  Does that mean lock channel 3?
if so, for the 0x70 device that's the equivalent of 'tmux[0][3].try_lock()'
I used a **similar syntax that worked** when creating devices: like 
'veml7700 = adafruit_veml7700.VEML7700(tmux[m][c])'

but I have tried various versions  of try_lock() in a nested loop for mux and channel and none work, :

```
'tmux[mux][channel].try_lock()' - this one just hangs - maybe that means its worked, but then what?.
'tmux[mux].try_lock()' - AttributeError: 'TCA9548A' object has no attribute 'try_lock'
'TCA9548A_Channel.tmux[mux][channel].try_lock()' - AttributeError: 'type' object has no attribute 'tmux'
'TCA9548A_Channel.tmux[mux].try_lock()' - AttributeError: 'type' object has no attribute 'tmux'
```

[tca example.txt](https://github.com/adafruit/Adafruit_CircuitPython_TCA9548A/files/6030874/tca.example.txt)

caternuson, 2021-02-23:
Not sure why you got that ` TypeError: can't convert str to int` error. It doesn't seem to match the code above it.

Here's a rough example, but this is very much an ugly hack:
```python
Adafruit CircuitPython 6.1.0 on 2021-01-21; Adafruit QT Py M0 with samd21e18
>>> import board
>>> import adafruit_tca9548a
>>> i2c = board.I2C()
>>> tca = adafruit_tca9548a.TCA9548A(i2c)
>>> i2c.try_lock()
True
>>> i2c.scan()
[112]
>>> i2c.unlock()
>>> tca[3].try_lock()
True
>>> i2c.scan()
[41, 112]
>>> tca[3].unlock()
>>> i2c.scan()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Function requires lock
>>>  
```

For the above, a VL53L0X was attached to output 3 of the TCA.

dretful, 2021-02-23:
**Thank you again Carter for your time on this.**
I copied your code, and made a program for my device, but slightly adjusted for the Pi Pico I am using where there is no default 'SDA' and 'SCL'.  
[hack.txt](https://github.com/adafruit/Adafruit_CircuitPython_TCA9548A/files/6032133/hack.txt)

I found some things out:

Firstly, 
When I run this for a **channel where there is nothing connected** (like 3) and **while the address is not explicitly stated**
`tca = adafruit_tca9548a.TCA9548A(i2c)`
 it runs through OK, but of course then no extra device is reported.
output:
```
i2c_lock True
ALL I2C addresses found: ['0x48', '0x4a', '0x70', '0x72']
i2c_unlock None
tca_lock True
TCA I2C addresses found: ['0x48', '0x4a', '0x70', '0x72']
tca_unlock None
Traceback (most recent call last):
  File "<stdin>", line 22, in <module>
RuntimeError: Function requires lock
```
When I choose a **channel where a device is installed on both of the TCA's**, I find that by default it looks at TCA at 0x70.  If there is no TCA at 0x70 it returns a null difference. 
```
i2c_lock True
ALL I2C addresses found: ['0x48', '0x4a', '0x72', '0x74']
i2c_unlock None
tca_lock True
TCA I2C addresses found: ['0x48', '0x4a', '0x72', '0x74']
tca_unlock None
Traceback (most recent call last):
  File "<stdin>", line 22, in <module>
RuntimeError: Function requires lock
```
 I suppose that's an expected result but I mention it.

Secondly,
When I **address the TCA specifically to one of the two that are installed** (or there IS a TCA at 0x70)
`tca = adafruit_tca9548a.TCA9548A(i2c, **address=0x72**)`
and scan for channel 0 **that does have something installed**
**BINGO!** a new device **0x10** is found, as it should be:
```
i2c_lock True
ALL I2C addresses found: ['0x48', '0x4a', '0x70', '0x72']
i2c_unlock None
tca_lock True
TCA I2C addresses found: ['0x10', '0x48', '0x4a', '0x70', '0x72']
tca_unlock None
Traceback (most recent call last):
  File "<stdin>", line 22, in <module>
RuntimeError: Function requires lock
```
And trying channel 2, with a different item plugged in with address 0x77 also works
```
i2c_lock True
ALL I2C addresses found: ['0x48', '0x4a', '0x70', '0x72']
i2c_unlock None
tca_lock True
TCA I2C addresses found: ['0x48', '0x4a', '0x70', '0x72', '0x77']
tca_unlock None
Traceback (most recent call last):
  File "<stdin>", line 22, in <module>
RuntimeError: Function requires lock
```
Trying a channel with nothing installed (1) gives the correct nil difference
```
i2c_lock True
ALL I2C addresses found: ['0x48', '0x4a', '0x70', '0x72']
i2c_unlock None
tca_lock True
TCA I2C addresses found: ['0x48', '0x4a', '0x70', '0x72']
tca_unlock None
Traceback (most recent call last):
  File "<stdin>", line 22, in <module>
RuntimeError: Function requires lock
```
Thirdly,
Checking the other TCA (0x70) on channel 0 however initially failed, with a hang.  I traced this to **two modules each of which in isolation caused it to hang, VEML6040 and VEML6075** (though I had been able to read from them when instantiated).  When I tried a BME280 of a type that had worked above on channel 1 (or any other) it worked fine. I think its these modules and the hanging they caused that confused me to some extent before.
So the good result after this was
```
i2c_lock True
ALL I2C addresses found: ['0x48', '0x4a', '0x70', '0x72']
i2c_unlock None
tca_lock True
TCA I2C addresses found: ['0x48', '0x4a', '0x70', '0x72', '0x76']
tca_unlock None
Traceback (most recent call last):
  File "<stdin>", line 22, in <module>
RuntimeError: Function requires lock
```
 Perhaps notable is that during one test while hanging, I removed the VEML6040 and it released the hang, **_and was identified_** (so perhaps something is locking on these modules because when left plugged in the only recourse seems to be is reboot. - is the timeout by default 255 secs?)

Finally, Lessons learned - Review of code since last time:
I wrote:
'tmux[mux][channel].try_lock()' - this one just hangs - maybe that means its worked, but then what?.'
Well this **was** working, but I had not unlocked the i2c bus beforehand, and I did not then go ahead and scan, and afterwards unlock. - 3 things wrong
So the operative part of the code now reads:
```
i2c.unlock()
print ("I2C bus unlocked")
for mux in range(len(tca)):
    if tca[mux]!= 0:
        print ('scanning', mux, hex(mux+112), tmux[mux])
        for channel in range(len(tca[mux])):
            print ('trying to lock',channel,tmux[mux][channel])
            print(tmux[mux][channel].try_lock())
            print("ALL I2C addresses found:", [hex(device_address) for device_address in i2c.scan()])
            print(tmux[mux][channel].unlock())
```
and works fine.
I'll tidy all this up now and post a final version, then close the issue.  Thank you again for your help.

caternuson, 2021-02-23:
While scanning is a bit sloppy with the current state of the library, simply using the TCA with the sensor libraries should be easy. Is there a reason you need to scan?

dretful, 2021-02-24:
My reasons for scanning are: 
a) to see what has been plugged in has been recognised (some devices it seems cant be) and has been attached as intended
b) to check that everything has later been instantiated in functional code to the right places and to guide that process
c) just to properly understand multiplexers and coding multiples, that there is an I2C bus which can be address contended and then there are 8 'slots' for each multiplexer added which don't suffer address clashes

I agree that I now know how to use the library to connect various devices to several multiplexers as well as directly to the bus and then use them

Well, I just completed my scanner code and am attaching it.
[scan tca9548.txt](https://github.com/adafruit/Adafruit_CircuitPython_TCA9548A/files/6032502/scan.tca9548.txt)
Output:
```
>>> %Run -c $EDITOR_CONTENT
I2C bus locked, scanning. 3 devices found. i2c scan complete, I2C bus unlocked
tca9548a multiplexer tmux[0] created on 0x70
tca9548a multiplexer tmux[2] created on 0x72
other I2C address(es) found: ['0x48'] could be >1 device on an address here
device found on tca9548a tmux[0] channel 0 0x77
device found on tca9548a tmux[0] channel 1 0x4a
device found on tca9548a tmux[2] channel 0 0x10
device found on tca9548a tmux[2] channel 2 0x76
>>> 
```
My outstanding issues are:
1: This doesn't check to see that the device found between 0x70 and 0x77 actually IS a tca9548a. locking a channel and scanning it seems to work fine for a BMP280! 
2: My VEML6040 and VEML6075 hang the scan.  I wonder why?
2a: I'm not yet able to discover my new TCS34725 but that needs more work from me.
3: With two ADS1115's connected directly to the i2c bus and using the option to instantiate them with different addresses, I cant read each of them (having followed the example syntax).  
    ads1 = ADS.ADS1115(i2c, address=0x48)
    chan1 = AnalogIn(ads1, ADS.P0)
    ads2 = ADS.ADS1115(i2c, address=0x49)#0x49
    chan2 = AnalogIn(ads2, ADS.P0)`
    print("ads1 {:>5}\t{:>5.3f}".format(chan1.value, chan1.voltage))
    print("ads2 {:>5}\t{:>5.3f}".format(chan2.value, chan2.voltage))`
output
`ads1  4673	0.584
Traceback (most recent call last):
  File "<stdin>", line 173, in <module>
  File "adafruit_ads1x15/analog_in.py", line 47, in value
  File "adafruit_ads1x15/ads1x15.py", line 127, in read
  File "adafruit_ads1x15/ads1x15.py", line 162, in _read
  File "adafruit_ads1x15/ads1x15.py", line 198, in _write_register
  File "adafruit_ads1x15/ads1x15.py", line 198, in _write_register
  File "adafruit_bus_device/i2c_device.py", line 84, in write
OSError: [Errno 19] Unsupported operation`
.. but that's a different issue really.

So do you agree this should be closed? if so go ahead.  Thank you again for your invaluable help.
Tony

caternuson, 2021-02-24:
OK, let's close this issue. It sounds like you've got things pretty much figured out for what you've got setup. But I think improving how scanning works is a good general issue, so I'm going to open a new separate one for that. Could just rename this one, but will probably be cleaner to have a new thread.

dretful, 2021-02-24:
OK.  I'll do that.  Maybe a scanning focus can look at the hanging issue for some  devices and the 2 identical devices direct on i2c too.

dretful, 2021-02-24:
Closed.  I have a way of reading the addresses of hardware on the i2c bus and several multiplexers now.

closed 2021-02-24 by dretful