adafruit/Adafruit_CircuitPython_TCA9548A #25: Issue with implementation of VL53L0X and TCA9548A multiplexer

kgdthomas, 2021-02-23:
kgdthomas commented 1 hour ago
I've been trying to work one vl53l0x with a TCA9548A. My code is as follows:

```
import time
import board
import busio
import adafruit_vl53l0x
import adafruit_tca9548a

# Create I2C bus as normal
i2c = busio.I2C(board.SCL, board.SDA)

# Create the TCA9548A object and give it the I2C bus
tca = adafruit_tca9548a.TCA9548A(i2c)

# For each sensor, create it using the TCA9548A channel instead of the I2C object
vl53a = adafruit_vl53l0x.VL53L0X(tca[0])
#vl53b = adafruit_vl53l0x.VL53L0X(tca[1])

#tsl2 = adafruit_tsl2591.TSL2591(tca[1])

# print range 1 per second
while True:
    print("Range: {0}mm".format(vl53a.range))
#    print("Range: {0}mm".format(vl53b.range))
    time.sleep(1.0)

# Optionally adjust the measurement timing budget to change speed and accuracy.
# See the example here for more details:
#   https://github.com/pololu/vl53l0x-arduino/blob/master/examples/Single/Single.ino
# For example a higher speed but less accurate timing budget of 20ms:
# Measured in microseconds
# 1 second = 1000 millisecond = 1000000 microsecond
# vl53.measurement_timing_budget = 20000
# Or a slower but more accurate timing budget of 200ms:
# vl53.measurement_timing_budget = 200000
# The default timing budget is 33ms, a good compromise of speed and accuracy.
```
but every attempt leads to the following error.

```
%Run test-i2c.py
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/adafruit_bus_device/i2c_device.py", line 154, in __probe_for_device
    self.i2c.writeto(self.device_address, b"")
  File "/home/pi/.local/lib/python3.7/site-packages/adafruit_tca9548a.py", line 67, in writeto
    return self.tca.i2c.writeto(address, buffer, **kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/busio.py", line 116, in writeto
    return self._i2c.writeto(address, buffer, stop=stop)
  File "/home/pi/.local/lib/python3.7/site-packages/adafruit_blinka/microcontroller/generic_linux/i2c.py", line 49, in writeto
    self._i2c_bus.write_bytes(address, buffer[start:end])
  File "/home/pi/.local/lib/python3.7/site-packages/Adafruit_PureIO/smbus.py", line 308, in write_bytes
    self._device.write(buf)
OSError: [Errno 121] Remote I/O error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/adafruit_bus_device/i2c_device.py", line 160, in __probe_for_device
    self.i2c.readfrom_into(self.device_address, result)
  File "/home/pi/.local/lib/python3.7/site-packages/adafruit_tca9548a.py", line 61, in readfrom_into
    return self.tca.i2c.readfrom_into(address, buffer, **kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/busio.py", line 106, in readfrom_into
    return self._i2c.readfrom_into(address, buffer, stop=stop)
  File "/home/pi/.local/lib/python3.7/site-packages/adafruit_blinka/microcontroller/generic_linux/i2c.py", line 56, in readfrom_into
    readin = self._i2c_bus.read_bytes(address, end - start)
  File "/home/pi/.local/lib/python3.7/site-packages/Adafruit_PureIO/smbus.py", line 179, in read_bytes
    return self._device.read(number)
OSError: [Errno 121] Remote I/O error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/Desktop/test-i2c.py", line 19, in <module>
    vl53l0x = adafruit_vl53l0x.VL53L0X(tca9548a[0])
  File "/usr/local/lib/python3.7/dist-packages/adafruit_vl53l0x.py", line 142, in __init__
    self._device = i2c_device.I2CDevice(i2c, address)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_bus_device/i2c_device.py", line 50, in __init__
    self.__probe_for_device()
  File "/usr/local/lib/python3.7/dist-packages/adafruit_bus_device/i2c_device.py", line 163, in __probe_for_device
    raise ValueError("No I2C device at address: 0x%x" % self.device_address)
ValueError: No I2C device at address: 0x29
```

caternuson, 2021-02-23:
That hardware combo should work fine. It may be a connection issue. Please post in the forums with photos of your setup:
https://forums.adafruit.com

```python
Adafruit CircuitPython 6.1.0 on 2021-01-21; Adafruit QT Py M0 with samd21e18
>>> import board
>>> import adafruit_tca9548a
>>> import adafruit_vl53l0x
>>> tca = adafruit_tca9548a.TCA9548A(board.I2C())
>>> vlx = adafruit_vl53l0x.VL53L0X(tca[0])
>>> vlx.range
252
>>> vlx.range
180
>>> vlx.range
130
>>> vlx.range
95
>>> 
```
[image]

caternuson, 2021-03-10:
Closing for now. Can't recreate issue.

closed 2021-03-10 by caternuson