adafruit/Adafruit_CircuitPython_TCA9548A #28: Allow manual muxing?

caternuson, 2021-03-25:
Should we provide some mechanism to manually set the TCA mux? For use cases where this library is being used in conjunction with I2C devices that do not have a CircuitPython driver.

Example:
https://forums.adafruit.com/viewtopic.php?f=60&t=177241

Maybe something like a new `channel` property?
```python
tca.channel = 3
```
Could probably add a new class variable to track this also, to provide a read of current channel.

ladyada, 2021-03-27:
yah im not against it!

tannewt, 2021-03-29:
I don't think you want a second way to change the channel because it'll lead to confusion when using both ways at the same time. Instead, they should just use the appropriate channel object like a busio.I2C for driverless use.

caternuson, 2021-05-25:
The current code only switches the mux via `try_lock()`, which works fine with `BusDevice` since it calls `try_lock()` as part of the context manager.
https://github.com/adafruit/Adafruit_CircuitPython_BusDevice/blob/c322a567512277cc94f152b7f359d7c011d51d96/adafruit_bus_device/i2c_device.py#L137

None of the other `busio.I2C`-like methods do this. So something like this would not do any mux changes:
```python
tca[1].writeto(stuff)
tca[2].writeto(stuff)
```
One would have to do this:
```python
tca[1].try_lock()
tca[1].writeto(stuff)
tca[1].unlock()
tca[2].try_lock()
tca[2].writeto(stuff)
tca[2].unlock()
```

For some 3rd party driver that does not use `BusDevice`, one would have to do this:
```python
sensor = SomeSensor(tca[1])
tca[1].try_lock()
sensor.read_value()
tca[1].unlock()
```

Could maybe add mux changing code to all the `busio.I2C` methods, like `writeto`? Track current channel and only actually mux change if needed, to reduce unnecessary I2C traffic.