adafruit/Adafruit_CircuitPython_TCA9548A #30: software reset

codenio, 2021-05-07:
By passing dedicated GPIO Pin to TCA9548A object, we could define software reset function within this library.
Is this a valid feature..?

Thoughts & Comments..?

jposada202020, 2021-05-08:
I took at the datasheet. then I came and read the proposed changes it was not obvious how to do this. I guess we are talking here to use 
[image]?
I think it will be a little in the weeds to use it, but an example will help illustrate the case

caternuson, 2021-05-24:
The dedicated GPIO pin being passed in is also wired to the RESET pin on the TCA9548A? And this would just abstract the toggling of that pin into some `reset()` function? Something like that could be supported by adding new kwarg to init that defaults to `None`, so there would be no change to current behavior.

Keep in mind you can do this already, by just toggling the GPIO pin in you user code.

codenio, 2021-05-25:
> Keep in mind you can do this already, by just toggling the GPIO pin in your user code.

yes, raised this issue exactly after the realization that It could actually be a feature. 

A Software handler for existing hardware reset functionality, which can be used for handling exceptions with i2c MUX

caternuson, 2023-11-10:
For anyone still monitoring this thread, there's a PR here that adds this:
https://github.com/adafruit/Adafruit_CircuitPython_TCA9548A/pull/50
Please test this with your setups if possible to see if it helps recover from the bad state.

caternuson, 2024-02-02:
Closing for now due to lack of response. A PR was made (#50) for this, but closed without merging since there was no feedback from this thread.

closed 2024-02-02 by caternuson