adafruit/Adafruit_CircuitPython_TCA9548A #31: Bug : Channel Switch Missing in TCA9548A_Channel.scan

codenio, 2021-05-07:
**Action:**  To perform i2c scan on a particular channel in TCA9548A  

```python
...
# Initialize I2C bus.
i2c = busio.I2C(I2C1_SCL_PIN,I2C1_SDA_PIN)
# default TCA address=0x70
tca = adafruit_tca9548a.TCA9548A(i2c)

print([hex(i) for i in i2c.scan()])
print([hex(i) for i in tca[0].scan()])
...
```

**Expected result:** Perform i2c scan on selected channel
```shell
$ python i2c-mux.py 
['0x70']
['0x60'] #MCP4725 DAC module is connected to channel 0 of TCA9548A
Press Entre to exit..
```
**Actual Result:** It Performs i2c scan on actual i2c bus where TCA9548A is connected
```
$ python i2c-mux.py 
['0x70']
['0x70']
Press Entre to exit..
```

Correction : TCA9548A_Channel.scan needs to be modified
https://github.com/adafruit/Adafruit_CircuitPython_TCA9548A/blob/e79e116856f153b856440384e90d9be5cd3f6fab/adafruit_tca9548a.py#L78-L80

caternuson, 2021-05-24:
Try as shown here:
https://github.com/adafruit/Adafruit_CircuitPython_TCA9548A/pull/27#issue-590329223
Don't forget to unlock after the scan also.

caternuson, 2022-02-23:
Closing due to lack of response. The call to `try_lock` is needed to make the mux channel change.

closed 2022-02-23 by caternuson