adafruit/Adafruit_CircuitPython_TCA9548A #34: Add advanced example

jposada202020, 2021-05-25:
New example is desired to show the user how to add two or more sensors

caternuson, 2021-05-25:
These are sensors that use `BusDevice`?

jposada202020, 2021-05-25:
Yes indeed.

caternuson, 2021-05-25:
The current `simpletest` example shows this for two TSL2591s.

jposada202020, 2021-05-25:
Thanks closing this then 👍🏼

caternuson, 2021-05-26:
We could still update the example offerings though. Maybe something like this?

- rename current `tca9548a_simpletest.py` to `multi_sensor.py` and that become the example for that
- add a new `tca9548a_simpletest.py` which could be an i2c scan sweep example
- any other use cases that are confusing and an example would help clarify?

jposada202020, 2021-05-26:
Oh yes I love the idea, will work on that :)

caternuson, 2021-05-26:
OK, reopened and renamed. I think there was a good scan example in that other issue thread?

jposada202020, 2021-05-26:
yes there is, I did example for the fakerequests, i was thinking something similar
https://github.com/adafruit/Adafruit_CircuitPython_FakeRequests/blob/main/examples/fakerequests_advancedtest.py
what do you think?

caternuson, 2021-05-27:
Even simpler, something like:
```python
import board
import adafruit_tca9548a

tca = adafruit_tca9548a.TCA9548A(board.I2C())

for channel in range(8): 
    if tca[channel].try_lock():
        print('Channel {}:'.format(channel), end='')
        tca[channel].scan()
        tca[channel].unlock()
```
Weird - I thought I had dumped something like that over in that PR #32 thread, but not seeing it now. Oh well. Easy enough to rewrite. Output would look something like:
```
Channel 0:[112]
Channel 1:[38, 56, 112]
Channel 2:[112]
Channel 3:[112]
Channel 4:[112]
Channel 5:[94, 112, 119]
Channel 6:[112]
Channel 7:[112]
```

Could maybe make it a little fancier and do a list comp on the return to filter out the TCA's address.

jposada202020, 2021-05-27:
Will do

jerryneedell, 2021-05-27:
Can you output the addresses in Hex?

jposada202020, 2021-05-27:
yes:) we like hex in the I2C world :)

caternuson, 2021-05-27:
```python
import board
import adafruit_tca9548a

i2c = board.I2C()

tca = adafruit_tca9548a.TCA9548A(i2c)

for channel in range(8): 
    if tca[channel].try_lock():
        print('Channel {}:'.format(channel), end='')
        addresses = tca[channel].scan()
        print([hex(address) for address in addresses if address != 0x70]) 
        tca[channel].unlock()
 ```

```
Channel 0:[]
Channel 1:['0x26', '0x38']
Channel 2:[]
Channel 3:[]
Channel 4:[]
Channel 5:['0x5e', '0x77']
Channel 6:[]
Channel 7:[]
```

PR #35 (2021-05-27): adding_example_improving_docs
close #34 
Included the new example as discussed in the issue.

Other Docs improvements
- Sphinx inter mapping: remove reference to register library as it is not required
- Add example to the API
- Add Learning Guide
- Add black badge
- Add Bus_register and blinka to requirements and setup.py
- Typos and other minor changes
closed 2021-05-27 by jposada202020