adafruit/Adafruit_CircuitPython_TCA9548A #42: AttributeError: module 'adafruit_ssd1306' has no attribute 'SSD1306'

Stormtracker017, 2022-11-28:
Trying to interface three 0.96" 128x64 OLEDs to a RPI3B. When running the program, I get the following error:

`Traceback (most recent call last):
  File "/home/pi/Desktop/GPS Python Programs/Multiple Display/gpsoledmulti.py", line 48, in <module>
    tsl1 = adafruit_ssd1306.SSD1306(tca[0])
AttributeError: module 'adafruit_ssd1306' has no attribute 'SSD1306'`

`import serial

import board

from PIL import Image, ImageDraw, ImageFont

import adafruit_ssd1306

import adafruit_tca9548a

ser = serial.Serial("/dev/ttyS0", baudrate=9600, timeout=10)

i2c = board.I2C()  

tca = adafruit_tca9548a.TCA9548A(i2c)

tsl1 = adafruit_ssd1306.SSD1306(tca[0])

tsl2 = adafruit_ssd1306.SSD1306(tca[1])

tsl3 = adafruit_ssd1306.SSD1306(tca[2])`

Neradoc, 2022-11-28:
That's correct, the library doesn't have a SSD1306 class. You probably meant to use `SSD1306_I2C`.
See how to use the library in the docs (replacing the I2C parameter with the TCA channel).
https://docs.circuitpython.org/projects/ssd1306/en/latest/
Or this guide for example, which uses the same type of display:
https://learn.adafruit.com/adafruit-pioled-128x32-mini-oled-for-raspberry-pi/usage

Note: for support questions like this, preferably go to the Adafruit [discord](https://adafru.it/discord) or [support forums](https://forums.adafruit.com/).

Stormtracker017, 2022-11-30:
Thanks.  Issue was solved.

closed 2022-11-30 by Stormtracker017