adafruit/Adafruit_CircuitPython_TCA9548A #44: PCA9546 not showing connected devices on ESP32S3 and ESP32S2

dave-ct, 2023-01-29:
Feather: Adafruit Feather ESP32S3 4MB Flash 2MB PSRAM with ESP32S3
Board: PCA9546 4-Channel STEMMA QT Multiplexer
CircuitPython version: `8.0.0-beta.6 `(Issues oalso on 7.3)
Example following:  https://learn.adafruit.com/adafruit-pca9546-4-channel-stemma-qt-multiplexer/circuitpython-python

Description:  Device has been setup following the example above.  Also expereince the exact same issue on the ESP32-S2 Feather.   I am unable to see any devices connected to the PCA9546 using the receommnded adafruit_tca9548a library.  When running the exmaple code the output is as below and this is with or wihout devices connected to the PCA9546.  Have swapped cables and also tried to different PCA9546 boards and get the same results.  Any suggestions on how to correct this?
```
Channel 0:['0xb']
Channel 1:['0xb']
Channel 2:['0xb']
Channel 3:['0xb']
```

The code being run is per the example.  
```
# SPDX-FileCopyrightText: 2021 Carter Nelson for Adafruit Industries
# SPDX-License-Identifier: MIT

# This example shows using TCA9548A to perform a simple scan for connected devices
import board
import adafruit_tca9548a

# Create I2C bus as normal
# i2c = board.I2C()  # uses board.SCL and board.SDA
i2c = board.STEMMA_I2C()  # For using the built-in STEMMA QT connector on a microcontroller

# Create the PCA9546A object and give it the I2C bus
mux = adafruit_tca9548a.PCA9546A(i2c)

for channel in range(4):
    if mux[channel].try_lock():
        print("Channel {}:".format(channel), end="")
        addresses = mux[channel].scan()
        print([hex(address) for address in addresses if address != 0x70])
        mux[channel].unlock()
```

BlitzCityDIY, 2023-01-31:
hihi- i just tested on a feather esp32-s2 with the PCA9546 and then the PCA9548 (8 channel) and was getting similar results with the `0xb` address showing up, but the connected sensor addresses were showing up alongside it:
```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Channel 0:['0xb', '0x29']
Channel 1:['0xb', '0x29']
Channel 2:[]
Channel 3:['0xb']
```
`0xb` is the address of the onboard LC709203F battery monitor, so it seems like it's getting picked up by the code as it does a sweep for connected I2C addresses. I was able to read the STEMMA sensors though and run code for the sensors. Can you try running the [multi-sensor example code](https://learn.adafruit.com/adafruit-pca9546-4-channel-stemma-qt-multiplexer/circuitpython-python#multi-sensor-example-code-3134675) with your setup? Replacing the light sensors with whatever sensor(s) you're trying to use?

dave-ct, 2023-01-31:
@BlitzCityDIY I will give it another go and see what results I get and sahre here. I did try the multi sensor code but was getting errors on the mux for the connected sensors.

dave-ct, 2023-02-05:
@BlitzCityDIY have tried it again, this time it has all worked with two SCD4x sensors.  Not sure what I was doing wrong last time but can put it down to user error :-). Thanks for the pointers and confirming it was my mistake.   For future reference here is the code and the output I tried incase anyone else has an issue.

```
# SPDX-FileCopyrightText: 2021 Carter Nelson for Adafruit Industries
# SPDX-License-Identifier: MIT

# This example shows using TCA9548A to perform a simple scan for connected devices
import board
import adafruit_tca9548a
import time
import adafruit_scd4x

# Create I2C bus as normal
# i2c = board.I2C()  # uses board.SCL and board.SDA
i2c = board.STEMMA_I2C()  # For using the built-in STEMMA QT connector on a microcontroller

# Create the TCA9548A object and give it the I2C bus
tca = adafruit_tca9548a.TCA9548A(i2c)

for channel in range(4):
    if tca[channel].try_lock():
        print("Channel {}:".format(channel), end="")
        addresses = tca[channel].scan()
        print([hex(address) for address in addresses if address != 0x70])
        tca[channel].unlock()
        
scd4x1 = adafruit_scd4x.SCD4X(tca[0])
scd4x2 = adafruit_scd4x.SCD4X(tca[1])

scd4x1.start_periodic_measurement()
scd4x2.start_periodic_measurement()
        
while True:
    if scd4x1.data_ready:
        print(
            "scd4x1 Co2 ppm: %d\tTemp: %0.1f*C\tHum: %0.1f%%"
            % (scd4x1.CO2, scd4x1.temperature, scd4x1.relative_humidity))
    if scd4x2.data_ready:
        print(
            "scd4x2 Co2 ppm: %d\tTemp: %0.1f*C\tHum: %0.1f%%"
            % (scd4x2.CO2, scd4x2.temperature, scd4x2.relative_humidity))
       
        time.sleep(1)
```

Output

```
]0;🐍Wi-Fi: off | REPL | 8.0.0-beta.6\
Adafruit CircuitPython 8.0.0-beta.6 on 2022-12-21; Adafruit Feather ESP32S3 4MB Flash 2MB PSRAM with ESP32S3
>>> %Run -c $EDITOR_CONTENT
Channel 0:['0xb', '0x62']
Channel 1:['0xb', '0x62']
Channel 2:['0xb']
Channel 3:['0xb']
scd4x1 Co2 ppm: 1161	Temp: 24.8*C	Hum: 46.0%
scd4x2 Co2 ppm: 1215	Temp: 23.2*C	Hum: 51.5%
scd4x1 Co2 ppm: 1154	Temp: 24.4*C	Hum: 46.7%
scd4x2 Co2 ppm: 1211	Temp: 22.8*C	Hum: 52.3%
scd4x1 Co2 ppm: 1156	Temp: 24.1*C	Hum: 47.4%
scd4x2 Co2 ppm: 1201	Temp: 22.6*C	Hum: 53.0%
```

closed 2023-02-05 by dave-ct