adafruit/Adafruit_CircuitPython_TCA9548A #45: No matching distribution found for typing-extensions~=4.0

grafst, 2023-05-06:
```
pi@raspberrypi:~/Adafruit_CircuitPython_TCA9548A $ python3 --version
Python 3.5.3
pi@raspberrypi:~/Adafruit_CircuitPython_TCA9548A $ pip3 --version
pip 9.0.1 from /usr/lib/python3/dist-packages (python 3.5)
pi@raspberrypi:~/Adafruit_CircuitPython_TCA9548A $
pi@raspberrypi:~/Adafruit_CircuitPython_TCA9548A $
pi@raspberrypi:~/Adafruit_CircuitPython_TCA9548A $
pi@raspberrypi:~/Adafruit_CircuitPython_TCA9548A $ pip3 install -r requirements.txt
Collecting Adafruit-Blinka (from -r requirements.txt (line 5))
  Using cached https://www.piwheels.org/simple/adafruit-blinka/Adafruit_Blinka-5.13.1-py3-none-any.whl
Collecting adafruit-circuitpython-busdevice (from -r requirements.txt (line 6))
  Using cached https://www.piwheels.org/simple/adafruit-circuitpython-busdevice/adafruit_circuitpython_busdevice-5.2.4-py3-none-any.whl
Collecting adafruit-circuitpython-typing (from -r requirements.txt (line 7))
  Using cached https://www.piwheels.org/simple/adafruit-circuitpython-typing/adafruit_circuitpython_typing-1.9.2-py3-none-any.whl
Collecting typing-extensions~=4.0 (from -r requirements.txt (line 8))
  Could not find a version that satisfies the requirement typing-extensions~=4.0 (from -r requirements.txt (line 8)) (from versions: 3.6.2, 3.6.2.1, 3.6.5, 3.6.6, 3.7.2, 3.7.4, 3.7.4.1, 3.7.4.2, 3.7.4.3, 3.10.0.0, 3.10.0.1, 3.10.0.2)
No matching distribution found for typing-extensions~=4.0 (from -r requirements.txt (line 8))
```

python 3.5.3
pip 9.0.1

FoamyGuy, 2023-05-06:
@grafst I think python `3.7.0` is the lowest version supported for using libraries with CPython.

Mine is a little newer than that, and pip installs typing-extensions 4.5.0 for me:
```
❯ python --version
Python 3.10.6

~ 
❯ pip install typing-extensions~=4.0
Defaulting to user installation because normal site-packages is not writeable
Collecting typing-extensions~=4.0
  Using cached typing_extensions-4.5.0-py3-none-any.whl (27 kB)
Installing collected packages: typing-extensions
Successfully installed typing-extensions-4.5.0

```

Try using python 3.7+ if you can.

grafst, 2023-06-09:
compiling python 3.8 (since there is no package for it on raspbian 9) from scratch fixed the issue for me

caternuson, 2023-11-10:
Closing. Seems resolved using a newer CPython release.

closed 2023-11-10 by caternuson