adafruit/Adafruit_CircuitPython_TCA9548A #48: TCA9548A ON TCA9548A

forever80-dev, 2023-11-10:
Hello,

Does the library allow access to a TCA in cascade on another?

Regards

caternuson, 2023-11-10:
It should. In general, nothing special is needed. The host doesn't care about the physical layout of the downstream I2C bus. Everything on the I2C bus just needs a unique I2C address. As long as that is maintained, should be fine.

The main responsibility for keeping things valid will be in your user code. That is - making sure not to enable TCA channels in such a way that multiple same address devices are exposed.

Keep in mind the TCA address can be changed up to 8 different values. That way, up to 64 channels could be provided without any cascading. Some non-cascading information on using multiple muxers here:
https://learn.adafruit.com/working-with-multiple-i2c-devices/multiple-muxers

What is your use case for needing to do this?

forever80-dev, 2023-11-11:
The configuration of my cards means that I have more than 8 TCA multiplexers in total on one line.
So is the unique address considered on an i2c bus or on a path?
for example is it allowed to do
TCA1 (0x70):
Channel 0: TCA 0x71
TCA2 (0x71):
Channel 0: TCA 0x70

caternuson, 2023-11-11:
Actually, thinking about this more, it won't work for your scenario, since you are already using 8 total TCA's. There are only 8 total unique addresses that can be set for the TCA itself. So, since all 8 addresses are taken by the first "layer", there's no way to avoid an address conflict with the other TCA's that would then be attached by chaining.

The initial 8 TCA's are always seen on the I2C bus. So those 8 addresses can not show up anywhere else. So there's no way to assign a non-conflicting address to the chained TCA's.

This is a hardware issue, not a software issue.

You are trying to use more than 64 total same address I2C devices?

closed 2024-02-02 by caternuson