adafruit/Adafruit_CircuitPython_TCA9548A #49: possible conflict between main I2C and multiplexed I2C devices

garyk10, 2023-11-10:
I'm trying to have 17 of the AS7341 spectrometers working simultaneously, and their 0x39 address is not changeable.  I have 16 working properly, with 8 on a TCA9548A at 0x70 and another 8 on a TCA9548A at 0x71.  However, when I add a AS7341 directly to the main I2C line (and not on a multiplexer), then that's the only one that works properly, with the other 16 yielding values of a power of 2 (or a sum of two powers of 2).

With 16 sensors connected (on 2 multiplexers):
(104, 185, 226, 370, 1042, 1283, 1889, 884)
(100, 184, 240, 377, 997, 1368, 1874, 751)
(110, 182, 246, 373, 1049, 1422, 1871, 864)
(108, 191, 237, 353, 1050, 1307, 1842, 911)
(111, 182, 239, 357, 1017, 1265, 1725, 875)
(95, 195, 239, 374, 990, 1377, 1861, 673)
(103, 180, 234, 356, 1020, 1324, 1820, 932)
(87, 168, 235, 345, 986, 1312, 1776, 843)
(81, 154, 194, 375, 1000, 1327, 1441, 497)
(78, 143, 214, 378, 872, 1044, 1494, 501)
(102, 187, 237, 401, 923, 1139, 1678, 742)
(82, 170, 225, 437, 1046, 1465, 1314, 541)
(89, 198, 237, 421, 1014, 1173, 1829, 663)
(87, 203, 229, 426, 1018, 1315, 1888, 617)
(94, 185, 227, 336, 1083, 1243, 1700, 622)
(87, 131, 218, 357, 659, 1106, 1579, 828)
All values are correct and consistent.

With 17 connected (one on main I2C line):
(96, 153, 224, 256, 17, 1230, 1280, 628)
(100, 151, 224, 256, 800, 1094, 1282, 756)
(101, 149, 229, 257, 16, 1156, 1280, 596)
(100, 157, 228, 256, 16, 1024, 1280, 656)
(101, 149, 229, 256, 808, 1222, 1057, 612)
(85, 129, 228, 256, 784, 1094, 1313, 672)
(101, 146, 225, 256, 816, 1032, 1280, 672)
(84, 134, 225, 256, 784, 1029, 1024, 576)
(81, 155, 193, 384, 800, 1036, 1280, 208)
(69, 138, 196, 256, 784, 193, 1312, 224)
(96, 153, 224, 384, 784, 1089, 1024, 720)
(80, 132, 197, 384, 0, 1158, 1280, 528)
(80, 132, 224, 384, 824, 1094, 1281, 692)
(84, 132, 229, 384, 816, 1034, 1280, 624)
(96, 155, 229, 256, 0, 1220, 1056, 608)
(84, 130, 196, 256, 520, 1100, 1024, 564)
(117, 159, 229, 384, 824, 1229, 1313, 756)
Only this last 17th sensor yields correct values each time.

I found this closed issue that might refer to a problem that I don't know about.  Is it possible to turn off an address on the main I2C?  Or is there some other solution?

              You only use the mux address to change the output channel state. After that, it's I2C as usual. So there's a potential to have a conflict with multiple devices of same address on different muxes. The fix is to turn off all channels on muxes not being used:
https://learn.adafruit.com/adafruit-tca9548a-1-to-8-i2c-multiplexer-breakout/wiring-and-test#multiple-multplexers-4-13
to "hide" the conflicting sensors.

_Originally posted by @caternuson in https://github.com/adafruit/Adafruit_CircuitPython_TCA9548A/issues/9#issuecomment-507474513_

caternuson, 2023-11-10:
You'll need to put everything behind muxers. The last one connected directly is always seen on the I2C bus. So it will conflict with the others.

garyk10, 2023-11-10:
That settles that.  Thank you!

closed 2023-11-10 by garyk10