adafruit/Adafruit_CircuitPython_TCA9548A #59: try / except block fails on the wrong iteration

RasmusB, 2025-07-10:
### CircuitPython version and board name

```python
Adafruit CircuitPython 9.2.8 on 2025-05-28; Seeed XIAO nRF52840 Sense with nRF52840
```

### Code/REPL

```python
import board
import adafruit_tca9548a

i2c = board.I2C()

i2c.try_lock()
addresses = i2c.scan()
print([hex(address) for address in addresses])
i2c.unlock()

test_boards = []

for addr in range(0x70, 0x78):
    print(f"Attemping address {addr:#2x}")
    temp = adafruit_tca9548a.TCA9548A(i2c=i2c, address=addr)
    try:
        temp[0].try_lock()
        temp[0].unlock()
        test_boards.append(temp)
    except Exception as e:
        print(f"No board on address {addr:#2x}, skipping...({e})")
        continue

print(test_boards)
```

### Behavior

```
code.py output:
['0x70', '0x71']
Attemping address 0x70
Attemping address 0x71
Attemping address 0x72
No board on address 0x72, skipping...([Errno 19] No such device)
Attemping address 0x73
```

After printing `Attemping address 0x73`, **all output stops and the program is stuck**. I can interrupt it normally with Ctrl+C or similar.

### Description

I'm building a board that can (optionally) be daisy chained with up to 8 identical boards. Each board has a TCA9548a set to an unique address between `0x70` and `0x77`.

Since I don't know at runtime how many boards are connected, or what addresses they use I start my program with a scan. 

In this example, two boards are connected using addresses `0x70` and `0x71`. Those are detected fine as shown at the top of the output. 

The program correctly detects that there is no board with address `0x72`. But when it gets to the next address (`0x73`) it just stops. I'm expecting it to fail gracefully the same way `0x72` did?

todbot, 2025-07-10:
Try putting `    temp[0].unlock()` in the except-block. Does that help?

dhalbert, 2025-07-10:
#52 describes a different failure, also related to getting stuck on the locking.

RasmusB, 2025-07-10:
Aah... that does make sense, thank you for the explanation! I thought I was going crazy for a while.

I guess I'll just do it the other way around for now, just scan the range I care about and only attempt a connection to addresses that responded to the `i2c.scan()` then.

Again, thanks for helping me make sense of this issue.

closed 2025-07-10 by RasmusB

dhalbert, 2025-07-10:
I agree, it is very confusing, and shows up some issues in the library.